Entware/Entware #1078: [Package Request] Amneziawg-go

mezhgano, 2024-08-23:
**Describe the solution you'd like**
AmneziaWG is a contemporary version of the WireGuard protocol. It's a fork of WireGuard-Go and offers protection against detection by Deep Packet Inspection (DPI) systems. At the same time, it retains the simplified architecture and high performance of the original.

The precursor, WireGuard, is known for its efficiency but had issues with detection due to its distinctive packet signatures. AmneziaWG addresses this problem by employing advanced obfuscation methods, allowing its traffic to blend seamlessly with regular internet traffic. As a result, AmneziaWG maintains high performance while adding an extra layer of stealth, making it a superb choice for those seeking a fast and discreet VPN connection.

https://github.com/amnezia-vpn/amneziawg-go

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

zyxmon, 2024-08-25:
http://zyxnerd.zyxmon.org/files/amneziawg-tools_1.0.20240213-1_mipsel-3.4.ipk
http://zyxnerd.zyxmon.org/files/amneziawg-go_v0.2.12-1_mipsel-3.4.ipk

Please give more info on usage if the packages are OK

mezhgano, 2024-08-25:
Wow, thanks for such a quick reply
Sorry, i misunderstood last task, i doesn't have **mipselsf** device to test this, only **armv7sf**...
It is possible to adopt package from mipsel to arm without to much effort?
Anyway, will post link to this issue to Russian forum right now, for someone who own mipsel device to test it.

zyxmon, 2024-08-25:
http://zyxnerd.zyxmon.org/files/amneziawg-tools_1.0.20240213-1_armv7-3.2.ipk
и
http://zyxnerd.zyxmon.org/files/amneziawg-go_nohf_v0.2.12-1_armv7-3.2.ipk
или
http://zyxnerd.zyxmon.org/files/amneziawg-go_v0.2.12-1_armv7-3.2.ipk

Когда то давно, я проверял userspace вариант обычного wg используя это материал - https://lowendtalk.com/discussion/comment/3073060 (не думаю, что что0то сильно поменялось, кроме добавленных опций амнезии).

zyxmon, 2024-08-26:
ip-full package is a missing dependency! Please install. On my mipsel device amneziawg-go creates wg0 interface:

root@Keenetic_Ultra:/opt/tmp$ ./amneziawg-go wg0
┌──────────────────────────────────────────────────────────────┐
│                                                              │
│       Running amneziawg-go is not required because this      │
│       kernel has first class support for AmneziaWG. For      │
│       information on installing the kernel module,           │
│       please visit:                                          │
| https://github.com/amnezia-vpn/amneziawg-linux-kernel-module │
│                                                              │
└──────────────────────────────────────────────────────────────┘
root@Keenetic_Ultra:/opt/tmp$ ip a
....
39: wg0: <POINTOPOINT,MULTICAST,NOARP> mtu 1420 qdisc noop state DOWN group default qlen 500
    link/none
root@Keenetic_Ultra:/opt/tmp$ uname -a
Linux Keenetic_Ultra 4.9-ndm-5 #0 SMP Thu Jun 6 13:52:52 2024 mips GNU/Linux

JkktBkkt, 2024-08-26:
Package ip-full (4.4.0-10b) installed in root is up to date.
and ip a provides output that looks correct, but it doesn't have the awg-created device (because awg crashes)

zyxmon, 2024-08-26:
@JkktBkkt - make an issue here - https://github.com/amnezia-vpn/amneziawg-go/issues

JkktBkkt, 2024-08-26:
Unfortunately that's basically pointless. They've removed the sections about freebsd and arm after forking, indicating that in these people are on their own and those configurations are not supported. Frankly, it's obvious that such architectures as mips aren't their priority.

zyxmon, 2024-08-29:
I've test it on my mipsel router and it works. log is here - https://forum.keenetic.com/topic/18794-amneziawg-go-для-entware/

mezhgano, 2024-10-02:
Только сейчас выдался момент протестировать, роутер ASUS RT-AX56U, если правильно понимаю, не ставиться из-за слишком старой версии архитектуры роутера...

```bash
$ opkg install amneziawg-go_nohf_v0.2.12-1_armv7-3.2.ipk
Unknown package 'amneziawg-go_nohf'.
Collected errors:
 * pkg_hash_fetch_best_installation_candidate: Packages for amneziawg-go_nohf found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package amneziawg-go_nohf.
```

```bash
$ opkg print-architecture
arch all 100
arch armv7-2.6 160
```

P.S. Не знаю, может просто попробовать изменить требования к архитектуре в файле `control` ipk пакета, вдруг получиться... 🤔