Entware/Entware #1083: Update package "accel-ppp"

Tokarak, 2024-09-02:
The accel-ppp package (https://github.com/Entware/rtndev/tree/master/accel-ppp) is outdated by five years. It's pointing at an old fork: the new organisational fork is https://github.com/accel-ppp/accel-ppp.

There is one new release (1.13.0, from March 2024). There are some crucial bug-fixes since then (e.g https://github.com/accel-ppp/accel-ppp/issues/131); master branch is stable, ci-tested, and reviewed, so I suggest building off HEAD (call it 1.13.0-1). 

The patches need updating, but I don't know how to do that.

I can test on aarch64-3.10 and compile myself if necessary (I set up a builder image).

- [x] No similar package already added and/or this one got some new killer feature(s),
- [x] Package is in active development (at least, not abandoned for 5 years or more),
- [x] I'm ready to test added package on my platform [e.g. mipselsf-k3.4] and get a feedback about it.

Tokarak, 2024-09-02:
I found the OpenWRT developer documentation on patching, so I could try working on it.

Tokarak, 2024-09-04:
Waiting for upstream to merge https://github.com/accel-ppp/accel-ppp/pull/188

Tokarak, 2024-09-20:
For posterity, my refreshed patches can be found here: https://github.com/Tokarak/rtndev/tree/accel-ppp-update

ryzhovau, 2025-01-16:
> The accel-ppp package (https://github.com/Entware/rtndev/tree/master/accel-ppp) is outdated by five years. It's pointing at an old fork: the new organisational fork is https://github.com/accel-ppp/accel-ppp.

Thanks for pointing out the correct URL. The package has been updated.

closed 2025-01-16 by ryzhovau