Entware/Entware #1086: adguardhome-go v0.107.52-1 frontend crash

zd3sf, 2024-09-11:
**Describe the bug**
adguardhome-go v0.107.52-1: crash on launch. Fresh install or opkg update, it doesnt matter. Logs crash at "loading module: sessions.DB". reverting back to old version (v0.107.44-1) solves the issue. 

**Environment:**
 - Entware feed: [arm7sf-k3.2]
 - Device [Netgear R7800]
 - Firmware version [DD-WRT v3.0-r58362 std (09/09/24] K6.1]

zyxmon, 2024-09-11:
Cannot confirm.
```
/share/CACHEDEV1_DATA/.qpkg/Entware/root # opkg install adguardhome-go
Installing adguardhome-go (v0.107.52-1) to root...
Downloading http://bin.entware.net/armv7sf-k3.2/adguardhome-go_v0.107.52-1_armv7-3.2.ipk
Configuring adguardhome-go.
/share/CACHEDEV1_DATA/.qpkg/Entware/root # uname -a
Linux QNAP-D1 3.10.20-al-2.5.3 #976 SMP PREEMPT Wed Jan 31 05:56:57 CST 2024 armv7l GNU/Linux
/share/CACHEDEV1_DATA/.qpkg/Entware/root # AdGuardHome
2024/09/11 17:08:09.720761 [info] AdGuard Home, version
2024/09/11 17:08:09.811562 [info] This is the first time AdGuard Home is launched
2024/09/11 17:08:09.811739 [info] Checking if AdGuard Home has necessary permissions
2024/09/11 17:08:09.811894 [info] warning: cannot check capability cap_net_bind_service: invalid argument
2024/09/11 17:08:09.812813 [info] AdGuard Home can bind to port 53
2024/09/11 17:08:09.820440 [info] safesearch default: disabled
2024/09/11 17:08:09.823699 [info] Initializing auth module: /share/CACHEDEV1_DATA/.qpkg/Entware/bin/data/sessions.db
2024/09/11 17:08:09.826255 [info] auth: initialized.  users:0  sessions:0
2024/09/11 17:08:09.826455 [info] AdGuard Home updates are disabled
2024/09/11 17:08:09.826555 [info] web: initializing
2024/09/11 17:08:09.826769 [info] This is the first launch of AdGuard Home, redirecting everything to /install.html
2024/09/11 17:08:09.827147 [info] AdGuard Home is available at the following addresses:
2024/09/11 17:08:09.828065 [info] go to http://127.0.0.1:3000
2024/09/11 17:08:09.828240 [info] go to http://10.0.0.36:3000
2024/09/11 17:08:09.828342 [info] go to http://[2a02:2168:87bd:9100:265e:beff:fe09:40c9]:3000
2024/09/11 17:08:09.828424 [info] go to http://[fe80::265e:beff:fe09:40c9%eth0]:3000
^C2024/09/11 17:08:16.360600 [info] Received signal "interrupt"
2024/09/11 17:08:16.360681 [info] stopping AdGuard Home
2024/09/11 17:08:16.360718 [info] stopping http server...
2024/09/11 17:08:16.361202 [info] stopped http server
2024/09/11 17:08:16.361469 [info] stopped

```

zd3sf, 2024-09-11:
But did you actually go to the UI and tried to login? Thats were the issue is.

zd3sf, 2024-09-11:
okay thank you for checking!