Entware/Entware #1091: "Expect" utility crashes on aarch64-k3.10 (runs ok on mipselsf-k3.4)

banditto9, 2024-10-06:
Package: Expect
Entware feed: aarch64-k3.10
Device: Keenetic Peak (KN-2710)
Firmware version: 4.2.1 Preview (same result on stable 4.1.7)
Issue: "expect" utility can't be launched after summer OPKG/Entware update (previous version of "expect" worked ok in the same opkg/entware setup). When launched it gives different errors.  Tried to make clean install of the opkg/entware - same result. Thank you for your time and help!

~ # opkg install expect
Installing expect (5.45.4-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/expect_5.45.4-1_aarch64-3.10.ipk
Configuring expect.

(1st run):
~ # expect
Segmentation fault

~ # opkg info expect
Package: expect
Version: 5.45.4-1
Depends: libc, libssp, librt, libpthread, tcl
Status: install user installed
Section: utils
Architecture: aarch64-3.10
Size: 132289
Filename: expect_5.45.4-1_aarch64-3.10.ipk
Description: Expect is a tool for automating interactive applications such as telnet, ftp,
 etc. Expect can make easy all sorts of tasks that are prohibitively
 difficult with anything else.
Installed-Time: 1728233225

(2nd run):
~ # expect
unable to realloc 2144487744 bytes
Aborted

--------------------------------------------

 In comparsion, it runs with no issues on MIPS (Keenetic Viva KN-1910 running 4.2.1 Preview):

~ # opkg install expect
Installing expect (5.45.4-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/expect_5.45.4-1_mipsel-3.4.ipk
Configuring expect.

~ # opkg info expect
Package: expect
Version: 5.45.4-1
Depends: libc, libssp, librt, libpthread, tcl
Status: install user installed
Section: utils
Architecture: mipsel-3.4
Size: 121388
Filename: expect_5.45.4-1_mipsel-3.4.ipk
Description: Expect is a tool for automating interactive applications such as telnet, ftp,
 etc. Expect can make easy all sorts of tasks that are prohibitively
 difficult with anything else.
Installed-Time: 1728233622

~ # expect
expect1.1>

The-BB, 2024-10-06:
Please, try it
[tcl_8.6.11-3_aarch64-3.10.ipk](http://bin.entware.net/aarch64-k3.10/test/tcl_8.6.11-3_aarch64-3.10.ipk)
(add [patch](https://gitweb.gentoo.org/repo/gentoo.git/tree/dev-lang/tcl/files/tcl-8.6.8-conf.patch) and bump PKG_RELEASE)
[expect_5.45.4-2_aarch64-3.10.ipk](http://bin.entware.net/aarch64-k3.10/test/expect_5.45.4-2_aarch64-3.10.ipk)
(bump PKG_RELEASE only)

banditto9, 2024-10-07:
Thank you very much for your prompt response!

1) tcl_8.6.11-3_aarch64-3.10.ipk solved the issue with the current "expect" version

root@KNP231:/opt/home$ opkg install tcl_8.6.11-3_aarch64-3.10.ipk
Upgrading tcl on root from 8.6.11-2 to 8.6.11-3...
Configuring tcl.
root@KNP231:/opt/home$ expect
expect1.1> exit

2) new "expect" also runs with no issues:

root@KNP231:/opt/home$ opkg install expect_5.45.4-2_aarch64-3.10.ipk
Upgrading expect on root from 5.45.4-1 to 5.45.4-2...
Configuring expect.
root@KNP231:/opt/home$ expect
expect1.1>

closed 2025-01-16 by ryzhovau