Entware/Entware #1092: [Package Request] UNFS3 v 0.10.0 for aarch64

tuhajbej6, 2024-10-08:
So UNFS3  (https://github.com/unfs3/unfs3)  apprently is missing its latest major build.
Since using it in LibreElec is the only way to get nfs server as it is kernel-module independent in  that semi- read-only distro.  Yet it still has some serious bugs. 
Now I truly believe that problems from 0.9.23 are solved in 0.10.0.

I am ready to try to compile it in docker but maybe a bit more clues about it - howto import dependencies.

ryzhovau, 2024-10-08:
Can't confirm that, sorry.
```
BusyBox v1.36.1 (2024-02-22 20:34:43 UTC) built-in shell (ash)

~ # opkg update
Downloading http://bin.entware.net/aarch64-k3.10/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
Downloading http://bin.entware.net/aarch64-k3.10/keenetic/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/keendev
~ # opkg install unfs3
Installing unfs3 (0.9.23-20150511-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/unfs3_0.9.23-20150511-1_aarch64-3.10.ipk
Installing libtirpc (1.3.4-1) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libtirpc_1.3.4-1_aarch64-3.10.ipk
Installing libwrap (7.6-4) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libwrap_7.6-4_aarch64-3.10.ipk
Installing rpcbind (1.2.6-2) to root...
Downloading http://bin.entware.net/aarch64-k3.10/rpcbind_1.2.6-2_aarch64-3.10.ipk
Installing libacl (2.3.2-2) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libacl_2.3.2-2_aarch64-3.10.ipk
Configuring libacl.
Configuring libwrap.
Configuring libtirpc.
Configuring rpcbind.
Configuring unfs3.
~ # unfsd -h
UNFS3 unfsd 0.9.23 (C) 2009, Pascal Schmidt <unfs3-server@ewetel.net>
Usage: unfsd [options]
	-h          display this short option summary
	-u          use unprivileged port for services
	-d          do not detach from terminal
	-e <file>   file to use instead of /opt/etc/exports
	-E          disable exports list procedure
	-i <file>   write daemon pid to given file
	-n <port>   port to use for NFS service
	-m <port>   port to use for MOUNT service
	-t          TCP only, do not listen on UDP ports
	-p          do not register with the portmapper
	-s          single user mode
	-b          enable brute force file searching
	-l <addr>   bind to interface with specified address
	-r          report unreadable executables as readable
	-T          test exports file and exit
```

tuhajbej6, 2024-10-08:
Hi, I am requesting for a newer build of the unfs3. The one in entware repository is 0.9.23,
while there is 0.10.0 on github sources.

closed 2025-01-16 by ryzhovau