Entware/Entware #1094: perl warning does not show line number

dowy, 2024-10-13:
**Describe the bug**
[perl - 5.40.0-1]: When perl emits a warning about the code, it does not show the line-number, instead it shows `%`
For example:
```shell
Entware$ perl -Mwarnings -E 'say $x'
Name "main::x" used only once: possible typo at -e line %.
Use of uninitialized value $x in say at -e line %.
```
This has the side-effect of causing some CPAN libraries to fail tests because they are expecting a line-number to appear.

(On a gentoo system running perl 5.40.0, the same command shows the line-number as expected:
```shell
gentoo$ perl -Mwarnings -E 'say $x'
Name "main::x" used only once: possible typo at -e line 1.
Use of uninitialized value $x in say at -e line 1.
```
)

**Environment (please complete the following information):**
 - Entware feed: x64-k3.2
 - Device QNAP TS-251D NAS
 - Firmware version QTS 5.2.0.2860 (2024/08/17)

**perl.control**:
```
Package: perl
Version: 5.40.0-1
Depends: libc, libssp, librt, libpthread, libbsd, libpthread
Source: feeds/packages/lang/perl
SourceName: perl
License: GPL-1.0-or-later Artistic-1.0-Perl
LicenseFiles: Copying Artistic README
Section: lang
SourceDateEpoch: 1721572854
URL: http://www.perl.com/
CPE-ID: cpe:/a:perl:perl
Maintainer: Marcel Denia <naoir@gmx.net>, Philip Prindeville <philipp@redfish-solutions.com>
Architecture: x64-3.2
Installed-Size: 4157440
Description:  Perl is a stable, cross platform programming language.
 It is used for mission critical projects in the public and private sectors
 and is widely used to program web applications of all needs.
```

dowy, 2025-02-03:
This issue still exists after upgrading to `5.40.0-2`

firmwarecostum, 2025-03-09:
you can use version 5.28 as a solution

The-BB, 2025-04-25:
https://github.com/openwrt/packages/issues/25912