Entware/Entware #1102: qBittorrent emergency update to 5.0.1 please

OneCDOnly, 2024-11-02:
Hi guys, looks like qBittorrent has been running without proper certificate validation for years. 

[qBittorrent fixes flaw exposing users to MitM attacks for 14 years](https://www.bleepingcomputer.com/news/security/qbittorrent-fixes-flaw-exposing-users-to-mitm-attacks-for-14-years/)

Can you please update the version available via Entware to 5.0.1?

Thank you. 🤓

Limpi1978, 2024-11-02:
+1 to this request!

My favourite torrent site banned all qBittorrent versions below v5.0.1 because of this:
https://cybersecuritynews.com/qbittorrent-rce-vulnerability/

So now I can not seed any torrents on my Asus RT-AX86U (running Asuswrt-Merlin v3004.388.8_2)... :'-(

Limpi1978, 2025-01-06:
v5.0.3 would be even better! ;)

ryzhovau, 2025-01-09:
Sorry, guys. [4.6.7](https://github.com/qbittorrent/qBittorrent/releases/tag/release-4.6.7) is the last one with qt5 support. 
I'm not ready to port newer qt.

closed 2025-01-09 by ryzhovau

Limpi1978, 2025-01-25:
Thanks for upgrading it to 4.6.7!

You can port newer qt later this year?