Entware/Entware #1103: Locale Install fails - Fallback works

ToS0, 2024-11-10:
**Describe the bug**
When installing Entware execution of https://bin.entware.net/aarch64-k3.10/locales_2.27-9_aarch64-3.10.ipk fails with no error and reboot at step `Adding en_EN.UTF-8`.

According to https://forum.dd-wrt.com/phpBB2/viewtopic.php?p=1214464 I could remove it during installation with `mv /opt/bin/localedef.new /opt/bin/localedef.new.old`. Amazingly the installer fell back to installing http://bin.entware.net/other/locale-archive.2.27 instead, which did work just fine.

Not sure if this may help someone to fix the issue. Anyway, great stuff guys - love it!

**Environment:**
 - Entware feed: bin.entware.net
 - Device Asus RT-AX92U
 - Firmware version: asuswrt-merlin

rodomb, 2025-01-06:
I have the same problem on ax86s.

ryzhovau, 2025-01-16:
So, this workaround is working as expected. Thank you for the feedback.

closed 2025-01-16 by ryzhovau