Entware/Entware #1108: opkg check_data_file_clashes between libe2p and e2fsprogs preventing installation of mc

deekayw0n, 2024-11-21:
if e2fsprogs is installed, mc cannot*** be installed/upgraded as it depends on libe2p (1.47.0-2) which is also provided by e2fsprogs (1.47.0-2)
 
[mc]: installation fails if e2fsprogs is installed due to clash.

Suspect one resolution would be to remove /opt/lib/libe2p.so.2 from e2fsprogs and have e2fsprogs depend on libe2p as well

**Environment:**
 - Entware feed: x64-k3.2
 - Device: NAS QNAP|TS-559 PRO
 - Firmware version: 4.2.6

**Expected Results:**
\# opkg install mc
Installing mc (4.8.31-1) to root...
Downloading http://bin.entware.net/x64-k3.2/mc_4.8.31-1_x64-3.2.ipk
Configuring mc.

**Actual Results:**
\# opkg install mc
Installing mc (4.8.31-1) to root...
Downloading http://bin.entware.net/x64-k3.2/mc_4.8.31-1_x64-3.2.ipk
Installing libe2p (1.47.0-2) to root...
Downloading http://bin.entware.net/x64-k3.2/libe2p_1.47.0-2_x64-3.2.ipk
Collected errors:
 * check_data_file_clashes: Package libe2p wants to install file /opt/lib/libe2p.so.2
	But that file is already provided by package  * e2fsprogs
 * check_data_file_clashes: Package libe2p wants to install file /opt/lib/libe2p.so.2.3
	But that file is already provided by package  * e2fsprogs
 * opkg_install_cmd: Cannot install package mc.

*** can be installed if opkg is given --nodeps parameter

ryzhovau, 2024-11-23:
Can't reproduce it on x64 or aarch64:
```
# opkg install mc
# opkg list_installed | grep libe2p
libe2p - 1.47.0-2
# opkg install e2fsprogs
Installing e2fsprogs (1.47.0-2) to root...
Downloading http://bin.entware.net/aarch64-k3.10/e2fsprogs_1.47.0-2_aarch64-3.10.ipk
Installing libcomerr (1.47.0-2) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libcomerr_1.47.0-2_aarch64-3.10.ipk
Installing libss (1.47.0-2) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libss_1.47.0-2_aarch64-3.10.ipk
Installing libext2fs (1.47.0-2) to root...
Downloading http://bin.entware.net/aarch64-k3.10/libext2fs_1.47.0-2_aarch64-3.10.ipk
Configuring libcomerr.
Configuring libss.
Configuring libext2fs.
Configuring e2fsprogs.
#
…
Please, install Entware from scratch, I think you are trying to update too old one.

closed 2024-11-23 by ryzhovau

deekayw0n, 2024-11-25:
That did the trick - I had Entware-std-1.0.3a installed (old school cool!).  

I backed up my etc/ and package list, removed this version, and installed Entware-std-1.0.3 from MyQNAP (myqnap.org) repo, reinstalled packages, restored etc/, and viola!  

mc can now be installed, no issues.

Thanks!