Entware/Entware #1114: TSHARK, executable of the Wireshark package, does not read CLI arguments

hikwqedxw, 2024-12-08:
**Describe the bug**
[wireshark: wireshark_3.6.24-1_armv7-3.2.ipk] Issue:  The executable 'tshark' does not accept any arguments from the command line.  You do not get any errors of any type, no help(-h, or --help), no version info(-v or --version) no syntax errors even when you deliberately give erroneous arguments.  For instance you give "--zebra" as an argument/option you do not get a syntax error, you just get the "...This could be dangerous." message and the program hangs until you send  a 'Ctrl-C' command to stop the execution.
Here is screen shot of a SSH session:

[image]

**Environment (please complete the following information):**
 - Entware feed: [http://bin.entware.net/armv7sf-k3.2/]
 - Device:  [Linksys WRT1900AC v1]
 - Firmware version [ DD-WRT v3.0-r58785]

firmwarecostum, 2024-12-21:
test openwrt

```
root@OpenWrt:~# tshark --version
TShark (Wireshark) 4.4.2 (Git commit 7e61ffa5bef8).
Copyright 1998-2024 Gerald Combs <gerald@wireshark.org> and contributors.
Licensed under the terms of the GNU General Public License (version 2 or later).
This is free software; see the file named COPYING in the distribution. There is
NO WARRANTY; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
Compiled (64-bit) using GCC 13.3.0, with GLib 2.82.0, with libpcap (including
remote capture support), with POSIX capabilities (Linux), without libnl, with
zlib 1.3.1, without zlib-ng, with PCRE2, with Lua 5.4.6, with GnuTLS 3.8.5, with
Gcrypt 1.10.3, with Kerberos (MIT), with MaxMind, with nghttp2 1.63.0, with
nghttp3 1.5.0, with brotli, with LZ4, with Zstandard, without Snappy, with
libxml2 2.13.4, with libsmi 0.5.0, with binary plugins.
Running on Linux 6.6.63, with ARMv8 Processor rev 4 (v8l), with 1792 MB of
physical memory, with GLib 2.82.0, with libpcap 1.10.5 (with TPACKET_V3), with
zlib 1.3.1, with PCRE2 10.42 2022-12-11, with c-ares 1.33.1, with GnuTLS 3.8.5,
with Gcrypt 1.10.3, with nghttp2 1.63.0, with nghttp3 1.5.0, with brotli 1.1.0,
with LZ4 1.10.0, with Zstandard 1.5.6, with libsmi 0.5.0, with LC_TYPE=C.UTF-8,
binary plugins supported.
root@OpenWrt:~# tshark --zebra
tshark: unrecognized option: zebra
Usage: tshark [options] ...
Capture interface:
  -i <interface>, --interface <interface>
                           name or idx of interface (def: first non-loopback)
  -f <capture filter>      packet filter in libpcap filter syntax
  -s <snaplen>, --snapshot-length <snaplen>
                           packet snapshot length (def: appropriate maximum)
  -p, --no-promiscuous-mode
                           don't capture in promiscuous mode
  -I, --monitor-mode       capture in monitor mode, if available
  -B <buffer size>, --buffer-size <buffer size>
                           size of kernel buffer (def: 2MB)
  -y <link type>, --linktype <link type>
                           link layer type (def: first appropriate)
  --time-stamp-type <type> timestamp method for interface
  -D, --list-interfaces    print list of interfaces and exit
  -L, --list-data-link-types
                           print list of link-layer types of iface and exit
  --list-time-stamp-types  print list of timestamp types for iface and exit
  --update-interval        interval between updates with new packets (def: 100ms)
Capture stop conditions:
  -c <packet count>        stop after n packets (def: infinite)
  -a <autostop cond.> ..., --autostop <autostop cond.> ...
                           duration:NUM - stop after NUM seconds
                           filesize:NUM - stop this file after NUM KB
                              files:NUM - stop after NUM files
                            packets:NUM - stop after NUM packets
Capture output:
  -b <ringbuffer opt.> ..., --ring-buffer <ringbuffer opt.>
                           duration:NUM - switch to next file after NUM secs
                           filesize:NUM - switch to next file after NUM KB
                              files:NUM - ringbuffer: replace after NUM files
                            packets:NUM - switch to next file after NUM packets
                           interval:NUM - switch to next file when the time is
                                          an exact multiple of NUM secs
                         printname:FILE - print filename to FILE when written
                                          (can use 'stdout' or 'stderr')
RPCAP options:
  -A <user>:<password>     use RPCAP password authentication
Input file:
  -r <infile>, --read-file <infile>
                           set the filename to read from (or '-' for stdin)
Processing:
  -2                       perform a two-pass analysis
  -M <packet count>        perform session auto reset
  -R <read filter>, --read-filter <read filter>
                           packet Read filter in Wireshark display filter syntax
[... 94 lines cut ...]
                           (default: /tmp)
  --compress <type>        compress the output file using the type compression format
Diagnostic output:
  --log-level <level>      sets the active log level ("critical", "warning", etc.)
  --log-fatal <level>      sets level to abort the program ("critical" or "warning")
  --log-domains <[!]list>  comma-separated list of the active log domains
  --log-fatal-domains <list>
                           list of domains that cause the program to abort
  --log-debug <[!]list>    list of domains with "debug" level
  --log-noisy <[!]list>    list of domains with "noisy" level
  --log-file <path>        file to output messages to (in addition to stderr)
Miscellaneous:
  -h, --help               display this help and exit
  -v, --version            display version info and exit
  -o <name>:<value> ...    override preference setting
  -K <keytab>              keytab file to use for kerberos decryption
  -G [report]              dump one of several available reports and exit
                           default report="fields"
                           use "-G help" for more help
Dumpcap can benefit from an enabled BPF JIT compiler if available.
You might want to enable it by executing:
 "echo 1 > /proc/sys/net/core/bpf_jit_enable"
Note that this can make your system less secure!
root@OpenWrt:~#
```

hikwqedxw, 2024-12-22:
> From "firmwarecostum" ***@***.***>
> To "Entware/Entware" ***@***.***>
> Cc "hikwqedxw" ***@***.***>; "Author"
> Date 12/21/2024 6:06:18 AM
> Subject Re: [Entware/Entware] TSHARK, executable of the Wireshark
> package, does not read CLI arguments (Issue #1114)

The version I am getting from the bin.entware.net is WIRESHARK 3.6.24-1
Entware is installed on a WRT1900AC v1 running DD-WRT r58819.

No matter what I put behind tshark on the commad line it gets ignored 
and I receive the "Running as user "root" and group "root". This could 
be dangerous." as if it was going to capture something then just hangs 
until I send a <ctrl-c>. No error messages of any kind, no help, no 
version info, nothing else.
I have eliminated the /opt partition and started over from scratch with 
no change.
Suggestions?

-------------------------------------------------------------------------------------------------------------------------------------------.

***@***.***:/opt# opkg update
Downloading http://bin.entware.net/armv7sf-k3.2/Packages.gz
Updated list of available packages in /opt/var/opkg-lists/entware
***@***.***:/opt# opkg upgrade
***@***.***:/opt# opkg list-installed wireshark
wireshark - 3.6.24-1
***@***.***:/opt# uname -a
Linux cecil-sapiyu 6.1.119-rt28 #267 SMP Sun Dec  1 04:29:52 +07 2024 
armv7l DD-WRT
***@***.***:/opt# free
               total        used        free      shared  buff/cache   
available
Mem:         251476      104092       86600           0       60784      
  18208
Swap:        262140        3692      258448
***@***.***:/opt# tshark
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# tshark --version
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# tshark --help
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# tshark --list-interfaces
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# tshark -D
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# tshark The Quick Brown Fox Jumped
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# tshark 0123456789
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# tshark .........
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# tshark will not read this command.
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# tshark -print
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# tshark -i br0 -w capture001.txt
Running as user "root" and group "root". This could be dangerous.
^C
***@***.***:/opt# ls
bin         etc         generic.sh  home        lib         libexec     
root        sbin        share       tmp         usr         var
***@***.***:/opt#

ryzhovau, 2025-06-10:
Command line options now working.
```
~ # tshark --version
Running as user "root" and group "root". This could be dangerous.
TShark (Wireshark) 4.4.6 (Git commit 920939a40074).
```
Closing for now.