Entware/Entware #1121: rustdesk-server

ripsu, 2025-01-01:
http://bin.entware.net/mipselsf-k3.4/rustdesk-server_1.1.11-1-1_mipsel-3.4.ipk

[rustdesk-server]:  hbbs
[2025-01-01 08:32:16.378301 +03:00] INFO [src/common.rs:122] Private key comes from id_ed25519
[2025-01-01 08:32:16.378635 +03:00] INFO [src/rendezvous_server.rs:1191] Key: FnWSf5k+uHBR44nlszREkiTjw6GOa0igaiyoQmFYQls=
[2025-01-01 08:32:16.378833 +03:00] INFO [src/peer.rs:84] DB_URL=./db_v2.sqlite3
thread 'sqlx-sqlite-worker-0' panicked at /home/me/E/staging_dir/target-mipsel_mips32r2_glibc-2.27/host/share/cargo/registry/src/index.crates.io-6f17d22bba15001f/ahash-0.7.8/src/random_state.rs:106:47:
getrandom::getrandom() failed.: Error { os_error: 89, description: "Function not implemented" }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Aborted

**Environment:**
 - Entware feed: [mipselsf-k3.4]
 - Device [keenetic giga 2]
 - Firmware version [entware?]

The-BB, 2025-01-01:
need more info
```root@ZKUII: /opt/root # opkg install rustdesk-server
Installing rustdesk-server (1.1.11-1-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/rustdesk-server_1.1.11-1-1_mipsel-3.4.ipk
Configuring rustdesk-server.
root@ZKUII: /opt/root # hbbs
[2025-01-01 12:43:17.581221 +03:00] INFO [src/common.rs:122] Private key comes from id_ed25519
[2025-01-01 12:43:17.581380 +03:00] INFO [src/rendezvous_server.rs:1191] Key: P+B3f8PzhJ0Cd1jBePMOfPVnvBMOwZI9SJAD4rsesKQ=
[2025-01-01 12:43:17.581485 +03:00] INFO [src/peer.rs:84] DB_URL=./db_v2.sqlite3
[2025-01-01 12:43:17.590655 +03:00] INFO [src/rendezvous_server.rs:99] serial=0
[2025-01-01 12:43:17.590798 +03:00] INFO [src/common.rs:46] rendezvous-servers=[]
[2025-01-01 12:43:17.590843 +03:00] INFO [src/rendezvous_server.rs:101] Listening on tcp/udp :21116
[2025-01-01 12:43:17.590894 +03:00] INFO [src/rendezvous_server.rs:102] Listening on tcp :21115, extra port for NAT test
[2025-01-01 12:43:17.590945 +03:00] INFO [src/rendezvous_server.rs:103] Listening on websocket :21118
[2025-01-01 12:43:17.591145 +03:00] INFO [libs/hbb_common/src/udp.rs:35] Receive buf size of udp [::]:21116: Ok(155648)
[2025-01-01 12:43:17.591551 +03:00] INFO [src/rendezvous_server.rs:138] mask: None
[2025-01-01 12:43:17.591653 +03:00] INFO [src/rendezvous_server.rs:139] local-ip: ""
[2025-01-01 12:43:17.591769 +03:00] INFO [src/common.rs:46] relay-servers=[]
[2025-01-01 12:43:17.592218 +03:00] INFO [src/rendezvous_server.rs:153] ALWAYS_USE_RELAY=N
[2025-01-01 12:43:17.592472 +03:00] INFO [src/rendezvous_server.rs:185] Start
[2025-01-01 12:43:17.593406 +03:00] INFO [libs/hbb_common/src/udp.rs:35] Receive buf size of udp [::]:0: Ok(155648)
^C[2025-01-01 12:46:50.697849 +03:00] INFO [src/common.rs:176] signal interrupt
root@ZKUII: /opt/root # uname -r
4.9-ndm-5
root@ZKUII: /opt/root # ```

ripsu, 2025-01-01:
> need more info

```

/opt/share/rustdesk # opkg install rustdesk-server
Installing rustdesk-server (1.1.11-1-1) to root...
Downloading http://bin.entware.net/mipselsf-k3.4/rustdesk-server_1.1.11-1-1_mips                                                                                                                                                             el-3.4.ipk
Configuring rustdesk-server.
/opt/share/rustdesk # hbbs
[2025-01-01 13:05:20.221699 +03:00] INFO [src/common.rs:148] Private/public key                                                                                                                                                              written to id_ed25519/id_ed25519.pub
[2025-01-01 13:05:20.221928 +03:00] INFO [src/rendezvous_server.rs:1191] Key: 0F                                                                                                                                                             o+4cnajt76OCe4g5oUx4S12z5YQL2fqEQyddz6d0A=
[2025-01-01 13:05:20.222150 +03:00] INFO [src/peer.rs:84] DB_URL=./db_v2.sqlite3
thread 'sqlx-sqlite-worker-0' panicked at /home/me/E/staging_dir/target-mipsel_m                                                                                                                                                             ips32r2_glibc-2.27/host/share/cargo/registry/src/index.crates.io-6f17d22bba15001                                                                                                                                                             f/ahash-0.7.8/src/random_state.rs:106:47:
getrandom::getrandom() failed.: Error { os_error: 89, description: "Function not                                                                                                                                                              implemented" }
note: run with `RUST_BACKTRACE=1` environment variable to display a backtrace
Aborted
/opt/share/rustdesk # uname -r
3.4.113
/opt/share/rustdesk #

```

ripsu, 2025-01-03:
http://bin.entware.net/mipselsf-k3.4/test/rust/rustdesk-server_1.1.9-1_mipsel-3.4.ipk

worked

closed 2025-01-04 by ripsu