Entware/Entware #1128: [Package Request] q

butaford, 2025-01-17:
**Describe the solution you'd like**
Добавьте пожалуйста пакет [q](https://github.com/natesales/q/)

Kdig, который в репе, по сути умеет опрашивать, только dot. Пакет уже протестирован на mipsel и aarch64

[image]

- [X] No similar package already added and/or this one got some new killer feature(s),
- [X] Package is in active development (at least, not abandoned for 5 years or more),
- [X] I'm ready to test added package on my platform and get a feedback about it.

butaford, 2025-01-17:
[q_pkgs.tar.gz](https://github.com/user-attachments/files/18451537/q_pkgs.tar.gz)
[q_src.tar.gz](https://github.com/user-attachments/files/18451536/q_src.tar.gz)

closed 2025-08-08 by The-BB

ryzhovau, 2025-09-14:
Апнем при очередной синхронизации в течение месяца.