Entware/Entware #1130: [Bug] coreutils-shuf

butaford, 2025-01-18:
**Describe the bug**
[coreutils-shuf]:
 При вводе, например команды:
`shuf -e item1 item2 item3 item4 item5`
или
`printf "%s" "item1 item2 item3 item4 item5" | tr ' ' '\n' | shuf -n 1`
возникает ошибка:
`shuf: getrandom: Function not implemented`
**Environment (please complete the following information):**
 - mipselsf-k3.4, aarch64
 - KN-1811, KN-3812
 - 4.2.3

The-BB, 2025-01-18:
getrandom может много где вылезти - ядра старые...
Глобально, для autotools:
```
diff --git a/include/site/linux b/include/site/linux
index 029245dc..323931fb 100644
--- a/include/site/linux
+++ b/include/site/linux
@@ -77,6 +77,7 @@ ac_cv_func_malloc_0_nonnull=yes
 ac_cv_func_realloc_0_nonnull=yes
 # getrandom since Linux 3.17 & getentropy glibc 2.25
 ac_cv_func_getentropy=no
+ac_cv_func_getrandom=no
 ac_cv_have_decl_getentropy=no
 # fix Y2038 problem since Linux 5.4(?) & glibc 2.32
 ac_cv_sys_year2038_opts='none needed'
```

zyxmon, 2025-01-19:
Ошибка с 2022 года, как сейчас проверил запуская бинарник из архива

butaford, 2025-01-20:
> getrandom может много где вылезти - ядра старые...

Да. Решил заменить:
`printf "%s" "item1 item2 item3 item4 item5" | tr ' ' '\n' | sort -R | head -n1`
тоже самое

zyxmon, 2025-01-23:
Как временное решение - скачайте из папки archive пакет версии 8.x - там проблемы быть не должно.

butaford, 2025-01-23:
> Как временное решение - скачайте из папки archive пакет версии 8.x - там проблемы быть не должно.

Да, 8.32-6 работает

butaford, 2025-06-03:
Я так понимаю вопрос решён? Пакет 9.6-5 рабочий.

closed 2025-06-03 by ryzhovau