Entware/Entware #1137: Could you update dnscrypt-proxyV2 2.1.5-1 to 2.1.7

Alozaros, 2025-02-04:
Could you update dnscrypt-proxy v2 to the last one plz... 2.1.7
for ARM based devices there are _nohf_ and _normal_ could you update both to 2.1.7

current dnscrypt-proxv2 is:
https://bin.entware.net/armv7sf-k3.2/dnscrypt-proxy2_nohf_2.1.5-1_armv7-3.2.ipk

- [ ] No similar package already added and/or this one got some new killer feature(s),
- [ ] Package is in active development (at least, not abandoned for 5 years or more),

https://github.com/DNSCrypt/dnscrypt-proxy/commit/d469ad13962fc11b0b0ad8f032aa4425c008638f

Alozaros, 2025-06-07:
Hi guys and thanks for the recent updates....but the recent update for DNScrypt-proxy v2 to 2.1.8-1 is not the last one and produces 
an error **[quad9-dnscrypt-ip4-filter-pri] should upgrade to XChaCha20 for encryption**
I already asked the guy who maintain the DNScrypt-proxy v2 if 2.1.8 has the issue with this type of encryption...

Im happy to test the last version for ARM7 and ARM64 ---if you provide me the update...or just update to the last if possible...

ryzhovau, 2026-11-03:
dnscrypt-proxy2_2.1.20 sources applied, binaries will arrive soon.

closed 2026-11-03 by ryzhovau