Entware/Entware #1141: Great developers, how can I change the program from installing to /opt to something else, such as /etc?

MortalM, 2025-02-12:
my device is xiaomi-AX3000T and redmi-AX6000, it's a route system ,program can't install on /opt , becasue / is read only system, can we chose the install /opt to other place , e.g. to /etc
Great developer please help me! This question has been bothering me for half a year now
If the issue is resolved, I will proactively close this session

[image]

And i re-edited the alternative.sh file, changed /opt to /tmp/opt, and then after installing it, it reported an error, because the opkg command solidified the configuration file address to /opt/etc/opkg.conf, how can I change the configuration file address obtained by the opkg file to /tmp/opt/etc/opkg.conf?
 
[image]

ryzhovau, 2025-02-12:
We are not ready to provide personal support here. Sorry.
I've seen a similar question here https://github.com/Entware/Entware/discussions/1109

closed 2025-02-12 by ryzhovau

MortalM, 2025-02-13:
> We are not ready to provide personal support here. Sorry.我们还没有准备在这里提供个人支持。对不起。 I've seen a similar question here [#1109](https://github.com/Entware/Entware/discussions/1109)我在这里看过一个类似的问题 [#1109](https://github.com/Entware/Entware/discussions/1109)

Thank you for your efforts