Entware/Entware #1155: Ncurses.pc Not Found in PKG-Config Search Path Error Compiling Python 3.11.10 on Debian (Bullseye)

garycnew, 2025-05-15:
When attempting to compile Python 3.11.10 on Debian (Bullseye), the config.log shows that the readline package cannot be built as the "Package ncurses was not found in the pkg-config search path."

I've ensured that libncurses-dev is installed:

$ sudo dpkg -l | grep -i libncurses | grep -i dev
ii  libncurses-dev:amd64                   6.2+20201114-2+deb11u2           amd64        developer's libraries for ncurses
ii  libncurses5-dev:amd64                  6.2+20201114-2+deb11u2           amd64        transitional package for libncurses-dev
ii  libncursesw5-dev:amd64                 6.2+20201114-2+deb11u2           amd64        transitional package for libncurses-dev

Ncurses.pc is available:

$ sudo dpkg -L libncurses-dev | grep -i ncurses.pc
/usr/lib/x86_64-linux-gnu/pkgconfig/ncurses.pc

The config.log error message recommends including the ncurses.pc directory in the PKG_CONFIG_PATH environmental variable.

What's the best way to include the ncurses.pc directory in the PKG_CONFIG_PATH environmental variable for Entware? Should it be added via the Makefile, Environment, or some other method? Is this a Red Herring?

**Environment (please complete the following information):**
 - Entware Feed: armv7sf-k3.2
 - Device: Multiple Asus Routers
 - Firmware Version: Asuswrt-Merlin

ryzhovau, 2025-05-16:
May be docker would be a better option. Please, take a look at reproducible build environment [here](https://github.com/Entware/docker).

closed 2025-05-16 by ryzhovau

garycnew, 2025-05-31:
Hi @ryzhovau As recommended, I installed docker and built the Entware docker container using (Debian Bullseye), but the same ncurses.pc error occurs, during the Python 3.11.10 compilation. Which Docker Host Flavor do you use, when building Entware? Thanks, again.