Entware/Entware #1167: [curl] 8.12.1 - authentication doesn't work

alxwk, 2025-06-16:
**Describe the bug**
[curl-8.12.1] can not use authentication. For example:

```
# curl -v -u "{name}:{password} --basic https://ipv4.nsupdate.info/nic/update
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
> GET /nic/update HTTP/1.1
> Host: ipv4.nsupdate.info
> User-Agent: curl/8.12.1
> Accept: */*
> 
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
< HTTP/1.1 401 Unauthorized
< Server: nginx
< Date: Mon, 16 Jun 2025 13:47:00 GMT
< Content-Type: text/plain
< Content-Length: 7
< Connection: keep-alive
< WWW-Authenticate: Basic realm="authenticate to update DNS"
< X-Content-Type-Options: nosniff
< Referrer-Policy: same-origin
< Cross-Origin-Opener-Policy: same-origin
< X-Frame-Options: DENY
< Vary: Accept-Language, Cookie
< Content-Language: en
< 
> GET /nic/update HTTP/1.1
> Host: ipv4.nsupdate.info
> User-Agent: curl/8.12.1
> Accept: */*
> 
< HTTP/1.1 401 Unauthorized
< Server: nginx
< Date: Mon, 16 Jun 2025 13:47:00 GMT
< Content-Type: text/plain
< Content-Length: 7
< Connection: keep-alive
< WWW-Authenticate: Basic realm="authenticate to update DNS"
< X-Content-Type-Options: nosniff
< Referrer-Policy: same-origin
< Cross-Origin-Opener-Policy: same-origin
< X-Frame-Options: DENY
< Vary: Accept-Language, Cookie
< Content-Language: en
< 
badauth~ # 
```
While the version 8.10.1 works OK:
```
~ # curl -v -u "{name}:{password}" --basic https://ipv4.nsupdate.info/nic/update
* TLSv1.3 (OUT), TLS handshake, Client hello (1):
* TLSv1.3 (IN), TLS handshake, Server hello (2):
* TLSv1.3 (IN), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (IN), TLS handshake, Encrypted Extensions (8):
* TLSv1.3 (IN), TLS handshake, Certificate (11):
* TLSv1.3 (IN), TLS handshake, CERT verify (15):
* TLSv1.3 (IN), TLS handshake, Finished (20):
* TLSv1.3 (OUT), TLS change cipher, Change cipher spec (1):
* TLSv1.3 (OUT), TLS handshake, Finished (20):
> GET /nic/update HTTP/1.1
> Host: ipv4.nsupdate.info
> Authorization: Basic Ym9{...}Q==
> User-Agent: curl/8.10.1
> Accept: */*
> 
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
* TLSv1.3 (IN), TLS handshake, Newsession Ticket (4):
< HTTP/1.1 200 OK
< Server: nginx
< Date: Mon, 16 Jun 2025 13:50:16 GMT
< Content-Type: text/plain
< Content-Length: 20
< Connection: keep-alive
< X-Content-Type-Options: nosniff
< Referrer-Policy: same-origin
< Cross-Origin-Opener-Policy: same-origin
< X-Frame-Options: DENY
< Vary: Accept-Language, Cookie
< Content-Language: en
< Strict-Transport-Security: max-age=31536000;
< X-Content-Type-Options: nosniff
< X-XSS-Protection: 1; mode=block
< Expect-CT: enforce, max-age=21600
< 
nochg 176.***.***.167~ #
```
Though 8.12 works, if the authentication base64 token is specified directly in headers by using `-H`. Judging by the [upstream issue #16333](https://github.com/curl/curl/issues/16333), it can be a build problem.

**Environment (please complete the following information):**
 - Entware feed: [aarch64-k3.10]
 - Device [Keenetic Ultra KN-1811]
 - Firmware version [KeeneticOS 4.3.3]

alxwk, 2025-06-17:
So, there's a new [option](https://github.com/openwrt/packages/commit/43b9a37a6ef0c69e834cf2967a3796804b7e3a48) which is disabled by default, and needs to be enabled at build to keep previous behavior.

deekayw0n, 2025-08-31:
wowzers - had to dive a bit deep in the rabbit hole to get here.  This naturally impacts any package that uses libcurl for basic web authorization including transmission-remote, git-http, and syslog-ng whic I suspect is a large number.

In my case, I was finding the exact same version of transmission-remote authenticating fine from a docker container but getting "401 Unauthorized" when run directly from my NAS with Entware.

Good root cause analysis @alxwk !  

Looking forward to #1168 getting pushed out.