Entware/Entware #1197: Package request: janus-gateway (and its libnice dependency)

VectorKappa, 2025-09-30:
**Describe the solution you'd like**
Entware is used on many commercial 3D printers, and one opensource self-hosted solution for quality-of-life 3D printer supervision is Obico. Obico requires janus-gateway as part of webcam livestreaming and printing error detection.

The [moonraker-obico](https://github.com/TheSpaghettiDetective/moonraker-obico) project (the part that lives on your printer) provides precompiled packages for raspberry pi, however, many printers run different architectures - in my case it's mipsel

The janus-gateway project is located [here](https://github.com/meetecho/janus-gateway)

Janus also depends on libnice by the freedesktop organization: 
https://libnice.freedesktop.org/

**Steps taken:**
- I've spoken to the Obico project maintainers, they have used the following configuration of Janus: `./configure --disable-all-plugins --enable-plugin-streaming --disable-all-transports --enable-websockets  --enable-rest --disable-docs `
- I tried to dabble with making it by myself - i got (most of) the dependencies ready but I'm running into compile errors with the main package itself, and my OpenWRT knowledge ends here.

- [ x ] No similar package already added and/or this one got some new killer feature(s),
- [ x ] Package is in active development (at least, not abandoned for 5 years or more),
- [ x ] I'm ready to test added package on my platform [e.g. mipsel-3.4] and get a feedback about it.

The-BB, 2025-11-16:
http://bin.entware.net/mipselsf-k3.4/test/libnice_0.1.22-1_mipsel-3.4.ipk
http://bin.entware.net/mipselsf-k3.4/test/stunbdc_0.1.22-1_mipsel-3.4.ipk (optional; from libnice)
http://bin.entware.net/mipselsf-k3.4/test/stund_0.1.22-1_mipsel-3.4.ipk (optional; from libnice)
http://bin.entware.net/mipselsf-k3.4/test/janus-gateway_1.3.3-1_mipsel-3.4.ipk
Please report if the packages is OK.

ryzhovau, 2026-11-04:
No response.

closed 2026-11-04 by ryzhovau